beaker-project/beaker #51: [RFE] Enable to deploy Beaker server / lab-controller on non-x86-64 environment

StykMartin, 2020-07-15:
**Is your feature request related to a problem? Please describe.**
At this moment it is not possible to deploy Beaker in environment without x86_64 machines.

Beaker itself is completely `noarch` product, however, we use dependencies that can be found only on particular architectures.
For example `syslinux`. This package is packaged only to `x86-64` RHEL 7.

We should get rid of dependencies like this to enable the user to install Beaker.
If dependencies enable simpler onboarding then we should mark them as prerequisites for a given environment (e.g. install syslinux on x86-64).

**Describe the solution you'd like**
I can install Beaker on top of PPC64.

Honggang-LI, 2022-04-21:
> Beaker itself is completely `noarch` product, however, we use dependencies that can be found only on particular architectures. For example `syslinux`. This package is packaged only to `x86-64` RHEL 7.

I installed beaker lab `controller` and `client` on aarch64 virtual machine. To get rid of `syslinux` dependency, I built a dummy package in name `syslinux` to cheat `beaker-lab-controller-28.2-1.el7bkr.noarch`.

I can provision virtual machines with the aarch64 beaker lab controller. I did not test the beaker `server` with aarch64 yet.

If only `syslinux` is your trobule, you may can cheat Beaker as I did.

desnesn, 2022-08-12:
A couple of years ago I did try to run/port the beaker server/controller packages on ppc64le and I think that what gave me the most issues on the server was porting selenium? Can't remember ...

Eventually, I just gave up and was able to find a x86_64 server to use.

However, the right way to provide support on other arches would most likely involve also porting the beaker server/controller dependencies.