beaker-project/beaker #57: [RFE]beaker should fail the job when no requestd hardware available

skycastlelily2, 2020-08-04:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

I try to reserve a 'SAS' server with "bkr workflow-simple --arch=x86_64 --distro=Fedora-Rawhide-20200607.n.0 Server --hostrequire devices_driver=SAS  --task=/distribution/check-install --task=/distribution/reservesys",but find 
that beaker reserve a kvm host,I think beaker should fail the job or queue reservation on a 'SAS' server when there is no available 'SAS' server.
I try to clone the job with web,get the same result.
I've changed the value to 'SATA',it works as there is available 'SATA' server.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

skycastlelily2, 2020-08-05:
Any update here?This is a little urgent,thanks.
The situation is worse than I thought:
When I try to reserve a SCSI server,beaker reserve a "SATA" one all the time,even though there are two available SCSI servers:
https://beaker.engineering.redhat.com/recipes/8637491

Name | Arch | Vendor | Model | LoanedTo | Status | Type | Action | User
-- | -- | -- | -- | -- | -- | -- | -- | --
hpe-bl465cgen8-01.hpe1.lab.eng.bos.redhat.com | i386, x86_64 | HP | ProLiant BL465c Gen8 |   | Automated | Machine | Reserve Now |  
hpe-dl320g5-01.hpe2.lab.eng.bos.redhat.com | i386, x86_64 | HP | ProLiant DL320 G5 |   | Automated | Machine | Reserve Now

StykMartin, 2020-08-05:
```xml
      <element name="device">
        <a:documentation xml:lang="en">
          Selects systems by device. If operator is ==, selects systems which
          have at least one matching device. If operator is !=, selects systems
          which have no matching devices.
        </a:documentation>
          <ref name="equality_op" />
        <oneOrMore>
          <choice>
            <attribute name="type" />
            <attribute name="bus" />
            <attribute name="driver" />
            <attribute name="vendor_id" />
            <attribute name="device_id" />
            <attribute name="subsys_vendor_id" />
            <attribute name="subsys_device_id" />
            <attribute name="description" />
          </choice>
        </oneOrMore>
      </element>
```

Definition of how to operate with devices. So, in this case, you should do
```xml
      <hostRequires>
        <device op="like" driver="%scsi%"/>
        <system_type value="Machine"/>
      </hostRequires>
```

StykMartin, 2020-08-05:
```--hostrequire devices_driver=SAS``` Will only pass the value into the XML section.
There is no validation if you are right or not.

```
        <device op="like" driver="%youdefinitelydonthaveit%"/>
```
Results into ```
Recipe ID 8638421 does not match any systems ```. Please always follow the XML schema.

skycastlelily2, 2020-08-06:
Hi, 
 Thanks a lot for your reply,the following works ,and it will queue if there is no available server
```      <hostRequires>
        <device op="like" driver="%scsi%"/>
        <system_type value="Machine"/>
      </hostRequires>
```

closed 2020-08-06 by skycastlelily2