beaker-project/beaker #71: [RFE] increase rows per page on host queries

jbastian, 2020-09-10:
**Is your feature request related to a problem? Please describe.**
When searching for a system in Beaker, the results are limited to 15 rows.  This leaves a lot of unused white space (on my monitor at least), and it's painfully slow clicking from one page to the next to look at all the systems.

**Describe the solution you'd like**
Increase the rows per page and/or make it customizable.

**Describe alternatives you've considered**
The bkr command line tool can list 1000s of systems, but entering raw hostRequires XML on the command line is not user friendly.

jbastian, 2020-09-10:
Thanks to @wackrat for finding the relevant code:

https://github.com/beaker-project/beaker/blob/beaker-28.0/Server/bkr/server/controllers.py#L300

If we can adjust the `limit=20` here, and for the similar queries for "Free" and "Available" and so on, that would be nice.

StykMartin, 2020-09-11:
I believe that this change is out of the scope for LTS.

StykMartin, 2021-04-13:
So, the code behind it will be irrelevant. However, we will need to get pagination on the new Beaker UI.
@renanrodrigo

renanrodrigo, 2021-04-13:
Doable. PatternMAGICfly even has the components for that.