beaker-project/beaker #74: [BUG] Kerberos authentication fails on Python 3.9

xhernandez, 2020-11-12:
**Describe the bug**  
When kerberos authentication is attempted, an exception is thrown:

```$ bkr whoami
Traceback (most recent call last):
File "/usr/bin/bkr", line 33, in <module>
sys.exit(load_entry_point('beaker-client==28.0', 'console_scripts', 'bkr')())
File "/usr/lib/python3.9/site-packages/bkr/client/main.py", line 113, in main
return cmd.run(*cmd_args, **cmd_opts.__dict__)
File "/usr/lib/python3.9/site-packages/bkr/client/commands/cmd_whoami.py", line 57, in run
self.set_hub(**kwargs)
File "/usr/lib/python3.9/site-packages/bkr/client/__init__.py", line 84, in set_hub
self.container.set_hub(username, password, auto_login=self.requires_login,
File "/usr/lib/python3.9/site-packages/bkr/client/command.py", line 328, in set_hub
self.hub = HubProxy(conf=self.conf, auto_login=auto_login)
File "/usr/lib/python3.9/site-packages/bkr/common/hub.py", line 87, in __init__
self._login()
File "/usr/lib/python3.9/site-packages/bkr/common/hub.py", line 114, in _login
login_method()
File "/usr/lib/python3.9/site-packages/bkr/common/hub.py", line 206, in _login_krbv
req_enc = base64.encodestring(res.token)
AttributeError: module 'base64' has no attribute 'encodestring'
```

**Version-Release number**  

28

**To Reproduce**  
Steps to reproduce the behavior:
1. Configure and initialize kerberes with `kinit`
2. Configure `bkr` to connect to the desired hub
3. Run `bkr whoami`

**Actual behavior**  
Kerberos connection doesn't work.

**Expected behavior**  
Kerberos connection should succeed and the command should complete successfully.

**Additional context**  
base64 module of Python 3.9 doesn't support `encodestring` method. It should use `encodebytes`.

xhernandez, 2020-11-12:
I've submitted PR #75 to fix this.

renanrodrigo, 2020-11-16:
#75 is merged. Thanks for the contribution!

closed 2020-11-16 by renanrodrigo