beaker-project/beaker #83: [BUG] Do not use deprecated kernel boot arguments

jstodola, 2020-12-14:
**Description of problem:**
When installing Fedora-Rawhide (future Fedora 34) in Beaker, the following warnings are printed by the installer:

[    9.574010] dracut-cmdline[299]: Warning: 'ks=<URL>' 
[    9.584962] dracut-cmdline[299]: Warning: ks has been deprecated and will be removed. Please use inst.ks instead. 
[    9.596929] dracut-cmdline[299]: Warning: 'ksdevice=00:00:1A:1B:0B:8C' 
[    9.604937] dracut-cmdline[299]: Warning: ksdevice has been deprecated and will be removed. Please use inst.ks.device instead. 
[    9.637034] dracut-cmdline[299]: Warning: 'ksdevice=<MAC>' is deprecated. Supplied BOOTIF takes precedence. 
...
Boot argument 'leavebootorder' is deprecated and will be removed in the future. Please use 'inst.leavebootorder' instead.
...

Fedora change page:
https://fedoraproject.org/wiki/Changes/Ignore_Anaconda_kernel_boot_parameters_without_inst_prefix

Please, use the proper kernel command line arguments with the "inst." prefix instead of the deprecated arguments.

Boot arguments without the "inst." prefix are marked as deprecated since RHEL-7:
https://access.redhat.com/documentation/en-us/red_hat_enterprise_linux/7/html/installation_guide/chap-anaconda-boot-options#sect-boot-options-deprecated-removed

**Version-Release number of selected component (if applicable):**
Beaker 27.4

StykMartin, 2020-12-14:
So. My plan is to implement this change only for F34+ and RHEL9+ so we can reduce the chance that we will break current workflows.

jstodola, 2020-12-14:
F34+ and RHEL9+ sound good to me.

PR #91 (2021-02-15): Fix deprecated kernel options for newer distros
This fixes #83.

Changed it for RHEL9+ and Fedora34+
closed 2021-02-15 by renanrodrigo