beaker-project/beaker #88: [RFE] clarify external/local in bkr-watchdog-extend(1)

AloisMahdal, 2021-02-09:
**Is your feature request related to a problem? Please describe.**
Learning about `bkr watchdog-extend` I wanted to know whether the command will adjust only external watchdog or both external.

But I could not tell neither from the manpage nor from `--help`

**Describe the solution you'd like**

`--help` and/or manpage should mention the fact that this command adjusts external watchdog only.

**Describe alternatives you've considered**

I asked on redhat-internal `#beaker` and got answer from @jbastian in few hours. ;-) 

**Additional context**
[image]

StykMartin, 2021-02-10:
We have only one watchdog in **Beaker** itself 👀

AloisMahdal, 2021-02-14:
I see, that's what I suspected.

I don't know where I've picked this double-watchdog idea but looking through Beaker docs I can't say it's responsible; it barely mentions watchdog and when it does, it just calls it "watchdog".  Maybe beaker-results anbd the WUI -- `External Watchdog Expired` message, and occasional `localwatchdog`.

Feel free to close it looks like I'm the only one that would benefit from the clarification.  On the other hand, if more people could be confused maybe it would still make sense to shimmy in a "don't confuse" note or something like that.

AloisMahdal, 2021-02-14:
.. on a second look, [Architecture Guide][ag] does mention it:

  [ag]: https://beaker-project.org/docs/architecture-guide/capabilities.html

> Watchdog timers
> ---------------
>
> Low level operating system testing is prone to rendering a machine completely unresponsive, especially when testing experimental code. Accordingly, Beaker supports two levels of watchdog timer, one running as part of the test harness (called the “Local Watchdog”) and one running on the lab controller associated with the system running the recipe (called the “External Watchdog”).
>
> If the local watchdog times out, it will abort the current task and attempt to move on to the next one. If the external watchdog times out, it will abort the entire recipe. Tasks are able to adjust the watchdogs dynamically if they need more time, allowing the use of more aggressive default timeouts.

StykMartin, 2021-02-14:
Keeping this open.
Would be nice to get rid of LWD notes. It is out of the scope for Beaker itself. Test harness may have many watchdogs and it depends on implementation what you will see in Beaker.