beaker-project/beaker #105: Fedora 31 harness not up on beaker-project.org

shall1024, 2021-04-13:
There is not a version of the restraint harness for Fedora 31 available at [https://beaker-project.org/yum/harness/](https://beaker-project.org/yum/harness/), but we want to run tests with Fedora 31. 

Without this harness, running tasks like `reservesys` result in the job incorrectly reporting failures/not notifying users that their machine is reserved, even if the OS installed correctly, etc.

Even though Fedora 31 is a bit older, is there a version of restraint suitable for it? If so, where can we find it?

StykMartin, 2021-04-13:
Hello, there is no Fedora 31 available at https://beaker-project.org/yum/harness/ because Fedora 31 is already marked as the end of life.

I can still build it for you in the community builder, but you will need to store it somewhere else.

shall1024, 2021-04-13:
Yes, that sounds great. We will be able to store it locally on our server. 
Thank you very much.

StykMartin, 2021-04-13:
> Yes, that sounds great. We will be able to store it locally on our server.
> Thank you very much.

Which architectures do you need @shall1024 ?
Upstream removed F31 build targets, however Brew (RH Koji) should still have it. Can you take a look @renanrodrigo ?

renanrodrigo, 2021-04-13:
We have here the latest version of restraint built for F31, which is 0.3.1.
Do you need the latest 0.3.3, @shall1024 ?

shall1024, 2021-04-13:
@StykMartin @renanrodrigo , we only need the x86_64 architecture, as for the version of restraint, I don't believe we have a specific requirement, so 0.3.1 should be sufficient.

Thank you.

renanrodrigo, 2021-04-14:
@shall1024 
https://file.io/WVC7EYuyz0kh
(single download I guess)

closed 2021-04-14 by shall1024