beaker-project/beaker #126: [RFE] Support other distributions

nmorey, 2021-04-28:
**Is your feature request related to a problem? Please describe.**
OpenFabrics is planning to use Beaker to automated RDMA kernel testing and to provide a dev/test/debug cluster for the OpenFabrics members.
As SUSE is interested in taking part in this, we would need Beaker to support other distros then RH based.
SUSE would obviously contribute for the autoyast part.

**Describe the solution you'd like**
Be able to deploy non RH-based distributions using Beaker

JohnVillalovos, 2021-04-28:
That would be very cool to get SuSE support. I took a stab it in the past and the XML based autoyast confused me 😵

At the moment we are able to install Debian 10.X and Ubuntu 18.04 using Beaker.  Probably could also install 20.04 but I would have to research that more as they changed the installer but I think it is possible to still be able to use deb-install on 20.04, but not 100% sure.

StykMartin, 2021-04-28:
I wouldn't like to dive into Ampere provisioning here. That's definitely not supported directly by Beaker. You hacked it a lot. @JohnVillalovos

JohnVillalovos, 2021-04-28:
> I wouldn't like to dive into Ampere provisioning here. That's definitely not supported directly by Beaker. You hacked it a lot. @JohnVillalovos

I tried to upstream all of our work though. The infrastructure is there, though the documentation is missing to be sure 😟

I think the biggest thing missing is likely the "kickstart" template file. I don't think I submitted that as it is not generic at this point. It needs work to make it templatable like the standard kickstart templates.

nmorey, 2021-04-29:
> That would be very cool to get SuSE support. I took a stab it in the past and the XML based autoyast confused me dizzy_face
> 
 That's great to hear. I wouldn't be able to help you on Autoyast directly as I have very little experience with it myself, but I think we should be able to find someone at SUSE who knows a thing or two about it :)
 I'll knock on a few doors internally and try to find someone who can help

StykMartin, 2021-04-29:
Making this work natively in Beaker will be more complicated than Autoyast. I believe there is tooling that we can use at least for validation.

I would like to know which SuSE version is a baseline for integration and where we can get the copies. 
Will we need multiarch or x86_64 only?
Is there any metadata around the composes, as Fedora/RHEL/CentOS has.

nmorey, 2021-04-29:
> I would like to know which SuSE version is a baseline for integration and where we can get the copies.

Tumbleweed is probably the target to start with. All the SLES/Leap inherits from it. So once Tumbleweed works, at least all future SUSE release should do too.

> Will we need multiarch or x86_64 only?

At the moment x86_64 should be enough. I saw that you have an RFE for multiarch so we'll see then :)

> Is there any metadata around the composes, as Fedora/RHEL/CentOS has.
I don't know what that is so can't say.

desnesn, 2022-08-12:
@StykMartin @nmorey @JohnVillalovos 

It seems that Beaker is on a maintenance state nowadays (specially upstream); thus, as far as I know, new features like these will hardly be ever implemented.

Nonetheless, I don't know if this still in your radar, but as a side project in my company, since we had the necessity, we tackled non-RHEL support on Beaker on the past year.

In summary, among other things, ***we implemented SLES and Ubuntu support :-)***
The ***beaker dn fork*** is able to ***generate autoyasts and autoinstalls automatically***, just like Beaker does kickstarts by default.
Also, we ***tweaked Fedora*** installations to use upstream repos; and also ***enabled CentoStream*** to be fetched directly from the external repos (with an ugly hack).

I have built binaries of an open sourced version of these patches at:

* COPR repo: https://copr.fedorainfracloud.org/coprs/desnesn/beaker.dn/
* Code fork: https://github.com/desnesn/beaker.dn
* SRPMs: https://github.com/desnesn/beaker.dn.rpms

Furthermore, the COPR can be enabled with:
~~~
dnf copr enable desnesn/beaker.dn
~~~

If someone plans to use this, ***read the setup instructions carefully on the beaker.dn.rpms repo linked above, to really understand what was tweaked***, as well as to fetch the SLES/Ubuntu custom harness packages.

Needless to say, there is no guarantees whatsoever of this tweaked usage.

Enjoy \o/

PS: downside is that we mostly used this to provision ppc64le guests, thus for other arches, the autoyast and autoinstall will likely need patches.