beaker-project/beaker #141: RPM build - CentOS Stream 9

StykMartin, 2021-10-03:


stratakis, 2021-12-08:
nose's development has stalled upstream since 2016. It's better to move the project to pytest and the conversion should be simple enough.

As for mock, python includes it in the stdlib so there is no need to use the external library.

hroncok, 2021-12-15:
See also https://fedoraproject.org/wiki/Changes/DeprecatePythonMock and https://fedoraproject.org/wiki/Changes/DeprecateNose

StykMartin, 2021-12-15:
That's completely fine. We will fix it, but first it has to be tracked :)

rasibley, 2022-05-02:
hi @StykMartin ! hope your doing well :-) curious if there is any update on this issue or are we still blocked due to the deprecated packages ?

StykMartin, 2022-05-02:
Hello @rasibley. I haven't worked on this yet. However, if it is a blocker for you I can dedicate some time and fix the issue.

rasibley, 2022-05-02:
Thank you @StykMartin ! Yes a bit of blocker for us as we would like to enable upstream kernel testing against the (AutoSD) Automotive Stream Distribution, see https://sigs.centos.org/automotive/#automotive-stream-distribution for more details. Thank you for working on it ! :hugs:

StykMartin, 2022-05-07:
Just a quick update. It is being worked on this problem. I just need to land a couple of PRs to remove obstacles.

rasibley, 2022-05-11:
Thank you @StykMartin !

rasibley, 2022-05-18:
@StykMartin looks like it was merged ! Thank you, should we close the issue and Yariv's issue as well which seems to be a dup ? Also where can I find the c9s Restraint builds ?

StykMartin, 2022-05-19:
C9 chroot was broken lately quite often therefore I had no chance to properly build stuff. If everything goes well we will have everything resolved by this weekend.

Yarboa, 2022-05-19:
Thanks you @StykMartin

StykMartin, 2022-05-24:
```
[root@4e5d275f28a8 /]# cat /etc/centos-release
CentOS Stream release 9
[root@4e5d275f28a8 /]# wget -O /etc/yum.repos.d/beaker-client.repo     https://beaker-project.org/yum/beaker-client-CentOSStream.repo
[root@4e5d275f28a8 /]# yum install beaker-client
Last metadata expiration check: 0:02:08 ago on Tue May 24 17:28:41 2022.
Dependencies resolved.
===================================================================================
 Package                  Arch        Version             Repository          Size
===================================================================================
Installing:
 beaker-client            noarch      28.3-1.el9          beaker-client      195 k
```

The client is in place. Btw, the beaker-client will be part of EPEL-9 as well -https://koji.fedoraproject.org/koji/buildinfo?buildID=1969906

Restraint is on the way as well.

StykMartin, 2022-05-25:
Client done. Closing ticket. Harness will be done under #150

PR #152 (2022-05-17): Enable build on CentOS Stream 9
I cherry-picked commits from `develop` to align packit behavior.
Then I enable C9 and removed all obstacles.

This will be released under 28.5
Fixes #141
closed 2022-05-25 by StykMartin