beaker-project/beaker #144: A `beaker-import` of RHEL 8.5 fails to add AppStream and BaseOS repos.

JohnVillalovos, 2021-12-03:
When importing RHEL 8.5 using `beaker-import` it fails to add the `AppStream` and `BaseOS` repos. This causes the install of `restraint` to fail as there are no repos available to download the required `make` package.

StykMartin, 2021-12-06:
Sadly I can't work on this, because I have no clue how 8.5 metadata looks like. I guess someone from RH will need to take a look. I guess you can provide debug output at least

mdujava, 2021-12-20:
Hey @JohnVillalovos can you compare if your tree is similar to the Centos8 at http://mirror.centos.org/centos/8/BaseOS/x86_64/os/, when importing ` distrotrees/X#repos` is empty, even after `.treeinfo` contains `repository=.` and `repository = ../../../AppStream/x86_64/os/`

beaker-import -v shows that 
```
2021-12-20 09:52:58,542 root DEBUG 	Importer ComposeInfoLegacy does not match
2021-12-20 09:52:58,640 root DEBUG 	Importer ComposeInfo does not match
2021-12-20 09:52:59,017 root DEBUG 	Importer TreeInfoLegacy does not match
2021-12-20 09:52:59,268 root DEBUG 	Importer TreeInfoRhel5 does not match
2021-12-20 09:52:59,451 root DEBUG 	Importer TreeInfoFedora does not match
2021-12-20 09:52:59,698 root DEBUG 	Importer TreeInfoFedoraArm does not match
2021-12-20 09:52:59,878 root DEBUG 	Importer TreeInfoRhel6 does not match
2021-12-20 09:53:00,115 root DEBUG 	Importer TreeInfoRHVH4 does not match
2021-12-20 09:53:00,303 root DEBUG 	Importer TreeInfoRhel7 Matches
2021-12-20 09:53:00,303 root INFO Attempting to import: http://mirror.centos.org/centos/8/BaseOS/x86_64/os/
2021-12-20 09:53:00,670 root DEBUG no addon repos for http://mirror.centos.org/centos/8/BaseOS/x86_64/os/, No option 'variants' in section: 'variant-BaseOS'
2021-12-20 09:53:00,673 root DEBUG 
{'arch': 'x86_64',
 'arches': [],
 'images': [{'path': 'images/pxeboot/vmlinuz', 'type': 'kernel'},
            {'path': 'images/pxeboot/initrd.img', 'type': 'initrd'}],
 'kernel_options': None,
 'kernel_options_post': None,
 'ks_meta': None,
 'name': 'CentOS Linux 8',
 'osmajor': 'CentOSLinux8',
 'osminor': '0',
 'repos': [],
 'tags': [],
 'tree_build_time': '1636765449',
 'urls': ['http://mirror.centos.org/centos/8/BaseOS/x86_64/os/'],
 'variant': 'BaseOS'}
2021-12-20 09:53:00,735 root INFO CentOS Linux 8 BaseOS x86_64 added to beaker.

```

StykMartin, 2021-12-20:
@mdujava what you posted is really an issue in CentOS 8 compose. I had a huge discussion about that, but there was 0 willing to change anything, therefore, I aborted any effort. This shouldn't be the case for RHEL as they follow the same strategy for composing thru the whole 8.