beaker-project/beaker #146: Clients not available for Fedora 35

JohnVillalovos, 2021-12-21:
This location does not have a client for Fedora 35:

https://beaker-project.org/yum/client/

JohnVillalovos, 2022-04-28:
@StykMartin Any chance you can generate these files? Thanks.

StykMartin, 2022-05-02:
@JohnVillalovos just checking what is missing. Can you please share the list of RPMs you missing there?

JohnVillalovos, 2022-05-02:
> @JohnVillalovos just checking what is missing. Can you please share the list of RPMs you missing there?

Sure

```
rhts-*  I use the 'make bkradd' command
beaker-client
```

Basically the files that are in: https://beaker-project.org/yum/client/Fedora33/

Thanks!

StykMartin, 2022-05-02:
All files provided for Fedora 35
https://beaker-project.org/yum/client/Fedora35/

Closing now.

closed 2022-05-02 by StykMartin

JohnVillalovos, 2022-05-02:
Thanks @StykMartin !!