beaker-project/beaker #150: [RFE]  Add CentOSStream 9 Repo under yum/download

Yarboa, 2022-04-17:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

In case i would like to install the following on CentOSStream9 
beaker-client
beaker-harness

Need to use the following repos

https://beaker-project.org/yum/beaker-client-Fedora.repo
https://beaker-project.org/yum/beaker-harness-Fedora.repo

Please refer 
**Describe the solution you'd like**

https://beaker-project.org/yum/beaker-client-CentOSStream.repo
https://beaker-project.org/yum/beaker-harness-CentOSStream.repo

with the following rpms:
https://beaker-project.org/yum/client/CentOSStream
https://beaker-project.org/yum/harness/CentOSStream

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

Fedora packages are installed as a w/a but no CentOS automotive upstream project will allow to refer 
Fedora packages as a dependency to CS9 

**Additional context**
you can refer this  
https://gitlab.com/CentOS/automotive/sample-images

Yarboa, 2022-05-02:
@StykMartin I am interested in doing the work.
Tried to to reach in libera.chat, any guidelines for the build?

StykMartin, 2022-05-02:
Hi @Yarboa.
I can look into this. Just to confirm. You need C9 Beaker Client and C9 Harness (Client + Server)
Is that right?

Yarboa, 2022-05-03:
> Hi @Yarboa. I can look into this. Just to confirm. You need C9 Beaker Client (#141 I need to fix this first) and C9 Harness (Client + Server) Is that right?

Ack,

Yarboa, 2022-05-03:
@StykMartin 
Many of the package does not have rpms
I used pipenv and ran Misc/rpmbuild.sh -bb
under pipenv shell

pip install gssapi
pip install mock
pip install nose
pip install sphinx

I also found this:
https://www.linuxfixes.com/2022/04/solved-how-to-include-pip-packages-in.html

Do you think it worth trying?

StykMartin, 2022-05-03:
Yeah, I wouldn't do that for sure. I will allocate some time this week and I will deliver packages as we usually do.

Yarboa, 2022-05-03:
> Yeah, I wouldn't do that for sure. I will allocate some time this week and I will deliver packages as we usually do.

Highly appreciated

Yarboa, 2022-05-25:
@StykMartin I think you can close #141
And continue with harness here, WDYT?

Yarboa, 2022-05-31:
@StykMartin thanks for the work done here #141 
Is there any progress related to harness?

StykMartin, 2022-06-03:
Hi @Yarboa. Initial Restraint for C9Stream is there. I had to use upstream spec. Feel free to test it. I will work on some changes meanwhile.

closed 2022-06-14 by StykMartin