beaker-project/beaker #158: [RFE] Add test harnesses for Fedora 36 and RHEL 9

JSpewock, 2022-07-26:
**Is your feature request related to a problem? Please describe.**
Importing the Fedora 36 and RHEL 9 distros works fine and they appear in beaker but provisioning with them always fails because it couldn't find a test harness.

**Describe the solution you'd like**
It would be nice if there were officially supported test harnesses for these distros in the beaker-project.org/yum/harness repository so that you can simply run `beaker-repo-update` and import them automatically rather than having to search for unofficial harnesses that may be out there and point your repo update there instead. It says in the docs that you can point it at the harness-testing repo to get harnesses that are release candidates instead but neither Fedora 36 or RHEL 9 are in there either. Fedora 36 has been out for a little while now and 37 is on the way so it would be nice to be able to use them in future provisioning to test on newer distros.

**Describe alternatives you've considered**
It seems like you could substitute in another http hosted repository when using the `beaker-repo-update` command but this is undesirable as it depends on whoever may be hosting the repository to maintain it.

desnesn, 2022-08-12:
@JSpewock

As far as I knew, the harness packages were not available at https://beaker-project.org/yum/harness/ mostly because RHEL9 and F36 were not GA, thus the packages could not be released. Nevertheless, RHEL9 and F36 both GAed on May. 

It seems that Beaker is on a maintenance state nowadays, but it still has support inside Red Hat. Thus, I think it is just a matter of the request reaching the correct Beaker engineer at Red Hat.

Maybe @StykMartin or @cbouchar knows who to forward these requests nowadays?

StykMartin, 2022-08-12:
I was keeping all packages upstream up to date with OS releases. Lately, I wasn't able to find time to make a new build.
I will try to make builds available during the weekend.

JSpewock, 2022-09-01:
@StykMartin Any update on this? I still don't see the packages for those distros available

StykMartin, 2024-06-13:
Harness is available for the latest available distributions (FRawhide, F40, F39, CentOS Stream 9).

closed 2024-06-13 by StykMartin