beaker-project/beaker #167: [BUG] Git sub modules error and bootstrap issue

vijay94, 2022-11-14:
I tried setting up the beaker in my local and faced issues with git sub modules getting timed out and the bootstrap version which was used in this project is not in active development and fresh install threw so many less errors(undefined variables and mixins).

mdujava, 2022-11-23:
Hey,

if you are using develop branch you may need to backport https://github.com/beaker-project/beaker/commit/0e33d509e1862b631e8e49b8a2927f53751705ce to update from git:// to https:// as Github stoped hosting repost over unencrypted git protocol.

vijay94, 2022-11-23:
I made those changes in my local and tried. But still bootstrap now uses scss instead of less. And most of the libraries used in beaker have updated their folder structure itself. So I had to do so many changes in styles to run the application locally.