beaker-project/beaker #168: [BUG] lessc having issues with bootstrap file

JohnVillalovos, 2022-11-22:
Unsure why I started seeing these errors in the /var/log/beaker/server-errors.log file.

Doing the following fails. Which seems to be what Beaker is doing in the traceback below
```
/usr/share/bkr/server/assets# lessc style.less
ArgumentError: Error evaluating function `percentage`: argument must be a number in /usr/share/bkr/server/assets/bootstrap/less/variables.less on line 292, column 27:
291 // -------------------------
292 @fluidGridColumnWidth:    percentage(@gridColumnWidth/@gridRowWidth);
293 @fluidGridGutterWidth:    percentage(@gridGutterWidth/@gridRowWidth);
```

Saw this in the log file which led me to the above:
```
Nov 22 14:00:20 beaker beaker-server[33210]: 2022-11-22 14:00:20,284 bkr.server ERROR app.py:log_exception():L1423 Exception on /bkr/client/ [POST]
Nov 22 14:00:20 beaker beaker-server[33210]: Traceback (most recent call last):
Nov 22 14:00:20 beaker beaker-server[33210]:   File "/usr/lib/python2.7/site-packages/flask/app.py", line 1817, in wsgi_app
Nov 22 14:00:20 beaker beaker-server[33210]:     response = self.full_dispatch_request()
Nov 22 14:00:20 beaker beaker-server[33210]:   File "/usr/lib/python2.7/site-packages/flask/app.py", line 1470, in full_dispatch_request
Nov 22 14:00:20 beaker beaker-server[33210]:     self.try_trigger_before_first_request_functions()
Nov 22 14:00:20 beaker beaker-server[33210]:   File "/usr/lib/python2.7/site-packages/flask/app.py", line 1497, in try_trigger_before_first_request_functions
Nov 22 14:00:20 beaker beaker-server[33210]:     func()
Nov 22 14:00:20 beaker beaker-server[33210]:   File "/usr/lib/python2.7/site-packages/bkr/server/wsgi.py", line 120, in init
Nov 22 14:00:20 beaker beaker-server[33210]:     assets.build_assets()
Nov 22 14:00:20 beaker beaker-server[33210]:   File "/usr/lib/python2.7/site-packages/bkr/server/assets.py", line 170, in build_assets
Nov 22 14:00:20 beaker beaker-server[33210]:     bundle.build()
Nov 22 14:00:20 beaker beaker-server[33210]:   File "/usr/lib/python2.7/site-packages/webassets/bundle.py", line 567, in build
Nov 22 14:00:20 beaker beaker-server[33210]:     disable_cache=disable_cache))
Nov 22 14:00:20 beaker beaker-server[33210]:   File "/usr/lib/python2.7/site-packages/webassets/bundle.py", line 504, in _build
Nov 22 14:00:20 beaker beaker-server[33210]:     force, disable_cache=disable_cache, extra_filters=extra_filters)
Nov 22 14:00:20 beaker beaker-server[33210]:   File "/usr/lib/python2.7/site-packages/webassets/bundle.py", line 429, in _merge_and_apply
Nov 22 14:00:20 beaker beaker-server[33210]:     kwargs=item_data)
Nov 22 14:00:20 beaker beaker-server[33210]:   File "/usr/lib/python2.7/site-packages/webassets/merge.py", line 272, in apply
Nov 22 14:00:20 beaker beaker-server[33210]:     return self._wrap_cache(key, func)
Nov 22 14:00:20 beaker beaker-server[33210]:   File "/usr/lib/python2.7/site-packages/webassets/merge.py", line 219, in _wrap_cache
Nov 22 14:00:20 beaker beaker-server[33210]:     content = func().getvalue()
Nov 22 14:00:20 beaker beaker-server[33210]:   File "/usr/lib/python2.7/site-packages/webassets/merge.py", line 252, in func
Nov 22 14:00:20 beaker beaker-server[33210]:     getattr(filter, type)(data, out, **kwargs_final)
Nov 22 14:00:20 beaker beaker-server[33210]:   File "/usr/lib/python2.7/site-packages/webassets/filter/less.py", line 97, in input
Nov 22 14:00:20 beaker beaker-server[33210]:     self.subprocess(args, out, in_)
Nov 22 14:00:20 beaker beaker-server[33210]:   File "/usr/lib/python2.7/site-packages/webassets/filter/__init__.py", line 505, in subprocess
Nov 22 14:00:20 beaker beaker-server[33210]:     proc.returncode, stdout, stderr))
Nov 22 14:00:20 beaker beaker-server[33210]: FilterError: less: subprocess returned a non-success result code: 1, stdout=, stderr=#033[31mArgumentError: Error evaluating function `percentage`: argument must be a number#033[39m#033[31m in #033[39mbootstrap/less/variables.less#033[90m on line 292, column 27:#033[39m
Nov 22 14:00:20 beaker beaker-server[33210]: #033[90m291 // -------------------------#033[39m
Nov 22 14:00:20 beaker beaker-server[33210]: 292 @fluidGridColumnWidth:    #033[7m#033[31m#033[1mp#033[22mercentage(@gridColumnWidth/@gridRowWidth);#033[39m#033[27m
Nov 22 14:00:20 beaker beaker-server[33210]: #033[90m293 @fluidGridGutterWidth:    percentage(@gridGutterWidth/@gridRowWidth);#033[39m#033[0m#033[0m
```

JohnVillalovos, 2022-11-23:
I ran a CentOS 7 container using `podman`. Installed the beaker yum repository file and then installed `beaker-server`.

Running the same command works in the container. I then did a `yum update` on the container and it still worked. I'm confused...

StykMartin, 2022-11-23:
Hello John,

It is caused by nodejs-less rpm bump - https://github.com/beaker-project/beaker-project.org/issues/38

_and I'm gone again..._

JohnVillalovos, 2022-11-23:
I just figured out that the version of `nodejs-less` package I am using is coming from EPEL. I'm guessing that is the issue.

And also what @StykMartin says above.

commit 949e3ea6 (2023-12-26): Require nodejs-less version lower than 2.0

Resolve: #168
closed 2023-12-27 by StykMartin