beaker-project/beaker #169: Clients not available for Fedora 37

mh21, 2022-11-28:
FC37 has been released 🎉, but the Beaker client doesn't have a repo for it at https://beaker-project.org/yum/client/:

[image]

This is a similar issue as https://github.com/beaker-project/beaker/issues/146.

@StykMartin would that be something you could look into, or can we sent a PR somewhere to fix it?

StykMartin, 2022-11-28:
Hello @mh21. Restraint builds are manually synced to the upstream node. I need to create a new build and I store it there.
If chroot is ready it should be quite simple.

mh21, 2022-11-29:
I updated the title and description to clarify that this also affects the harness repos at https://beaker-project.org/yum/harness/.

mh21, 2022-12-12:
friendly ping @StykMartin @cbouchar

FC35 is the latest Fedora version that has client/harness repos, and is going EOL tomorrow 🙈

StykMartin, 2022-12-13:
Well, I guess It is about time to store new RPMs if FC35 is dead 🫣.
In the morning I created all builds, I need to sync them upstream.

StykMartin, 2022-12-16:
F36/F37 stored. I haven't run any smoke tests yet.

mh21, 2022-12-16:
> F36/F37 stored. I haven't run any smoke tests yet.

The client repo at https://beaker-project.org/yum/client/ is still missing them for FC37?

StykMartin, 2022-12-16:
Oh.. I completely forgot Beaker Client. Is it really necessary to store it for FC there? I'm always building it directly to Fedora, but I can sync it

mh21, 2022-12-16:
> Oh.. I completely forgot Beaker Client. Is it really necessary to store it for FC there? I'm always building it directly to Fedora, but I can sync it

Sorry I didn't know that 🤦. If the official Fedora repos are the canonical way to get the client for Fedora, we should totally switch to that. Maybe it makes sense to update https://beaker-project.org/download.html with that info 🤗?

StykMartin, 2022-12-16:
I'm keeping a client here - https://src.fedoraproject.org/rpms/beaker (part of FC). As you can see it is always up-to-date. 
I will reconsider if there is any use case now to use BP.org. From the top of my mind, I don't see any right now.

I'll think about that and update BP.org accordingly.

mh21, 2022-12-20:
Thank you! From our side, the ticket is resolved. I'll leave it open so you can decide how to track the beakerproject.org changes 🤗

StykMartin, 2022-12-20:
Tracked here - https://github.com/beaker-project/beaker-project.org/issues/39. 

Closing. Thank you @mh21 !

closed 2022-12-20 by StykMartin