beaker-project/beaker #176: Harness not available for Fedora 38

tales-aparecida, 2023-08-31:
FC38 was released on April 🎉 , and the Beaker harness don't have a repo for it at https://beaker-project.org/yum/harness/

This is a similar issue as https://github.com/beaker-project/beaker/issues/169 and https://github.com/beaker-project/beaker/issues/158

@StykMartin would that be something you could look into, or can we sent a PR somewhere to fix it?

StykMartin, 2023-09-01:
Hello, I will build it during the weekend and ship it to bp.org

mh21, 2023-09-01:
Would it also be possible to already provide the harness for FC39 (which is now in [Beta freeze](https://fedorapeople.org/groups/schedule/f-39/f-39-key-tasks.html)) as well?

StykMartin, 2023-09-04:
So over the weekend I was able to rebuild all the targets (active Fedora, ELN, etc..), but I had to hack the build (again). I will backport the patches back to Restraint so this can be streamlined a bit. Sorry for the delay. RPMs will be released today.

StykMartin, 2023-09-04:
The harness repositories for F38 and F39 are up. I ran smoke tests on both, everything seems fine. If you can confirm that everything is working as expected, I will close this. However, major maintenance is needed on the harness itself :\.

closed 2023-09-08 by StykMartin