beaker-project/beaker #178: [BUG] New release

p3ck, 2023-09-29:
**Describe the bug**  
Who can publish a new release?

**Version-Release number**  
28.3 -> 28.4

StykMartin, 2023-09-29:
I'm currently maintaining all the upstream packages (including Fedora packages).
If you need we can create a new release.

p3ck, 2023-09-29:
Hi @StykMartin ,

I would be very thankful if you could spin a new release.  We currently use beaker in dci and the following PR's were reviewed and merged recently:

https://github.com/beaker-project/beaker/pull/177
https://github.com/beaker-project/beaker/pull/165

The first one is pretty simple and no real risk.  Only affect junit results on python3.
The second one is bigger but we've been running with it for almost a year now.  It fixes the issue of distro specific boot loaders, but only when providing the distro via the xml and not importing it.

Thanks!
Bill

StykMartin, 2023-10-01:
Hi @p3ck,

- The changes in #177 are straightforward, and there is no problem shipping it in a new minor release.
- #165 can be tricky because of the alembic changes. We would have different `HEAD` revisions for the same release, something that hasn't really happened in Beaker before. But I see no reason why we shouldn't make a new release 29.

StykMartin, 2023-10-12:
I'll cherry-pick #177 to release the branch and create a release from it.
PR #165 will remain on develop and we will not ship it in close future.

StykMartin, 2024-01-17:
It is done. CI is building 29.0

closed 2024-01-17 by StykMartin