beaker-project/beaker #192: `restraint-rhts` is the default harness but not available for Fedora 38 or Fedora 39.

JohnVillalovos, 2023-12-08:
It seems like `restraint-rhts` is the default harness in Beaker.  https://github.com/beaker-project/beaker/blob/601a4713c8e366e918bbeabe87745ef1e2d4854d/Server/bkr/server/model/distrolibrary.py#L64-L66

But Fedora 35 is the last harness directory where I see a `restraint-rhts` package.  https://beaker-project.org/yum/harness/Fedora35/

In the Fedora 39 harness directory I don't see it: https://beaker-project.org/yum/harness/Fedora39/

Should I specify a different harness? Should we change the default harness to something else?

Thanks.

StykMartin, 2023-12-09:
`restraint-rhts` is the safest option to run with for Beaker standpoint. 
If we would like to change it we should make sure that all our base tasks can work with replacement without issues.

Regarding the availability of this package - this is going to change - we will ship for the time being. The community is asking to provide -rhts https://github.com/restraint-harness/restraint/issues/306. I just need to resolve this https://github.com/restraint-harness/restraint/issues/307. Or we will change the restraint build process completely.

StykMartin, 2023-12-09:
Double checking and I can provide the latest restraint and rhts for Fedora w/ static build

StykMartin, 2024-06-13:
Aaand it is done.

closed 2024-06-13 by StykMartin