beaker-project/beaker #193: [RFE] Create OpenAPI Spec

StykMartin, 2023-12-21:
**Is your feature request related to a problem? Please describe.**
The Beaker Server and Beaker Labcontroller API are not properly documented.

**Describe the solution you'd like**
Create [OpenAPI Spec](https://swagger.io/specification/) for Beaker Server and Beaker Labcontroller. 
This will be the source of truth for implementation.

**Describe alternatives you've considered**
Avoid writing documentation, but I felt like people might not like it.

StykMartin, 2023-12-21:
Based on the discussion we had, I think you might want to take a look at this @p3ck

mdujava, 2023-12-21:
Some generated documentation can be accessed on `http://lab-controller-fqdn:8000/server`.
It would be ideal to get openapi spec generated from xmlrpc/turbogears directly, but few minutes searching on web did not reveal any out-of-the-box tool for it.

StykMartin, 2023-12-21:
The goal is to get rid of XMLRPC and Turbogears, so tooling is not necessary in this case.

StykMartin, 2023-12-21:
But good point @mdujava, I updated the description to make that clear.

mdujava, 2023-12-21:
Yeah, I would say when creating new REST api that would preserve functionality, would be easier and less problems would be introduced when existing API would be somehow defined (eg. list of all methods/endpoints).