beaker-project/beaker #200: [BUG] Change urlparse to use scheme='' instead of scheme=None

jonmasters, 2024-01-02:
There's a bug in the current beaker release when running on latest fully updated RHEL7. A security bugfix added to RHEL7 (which I think is actually broken because it changes behavior of existing code) adds a check for bad bytes in urlparse, causing it to now attempt to deference the url object as a string (though it might be None). The result is that distro imports are broken. The fix is to change distrotrees:add_distro_urls to use urlparse='' instead.

StykMartin, 2024-01-02:
Hello @jonmasters,

I'm aware of this issue - I completely agree that the bugfix is broken because it completely breaks the previous behavior. 

Anyway, I have fixed it in c6e4ef5af4ff17f54bcd189149e13eec7d50b84e. We are about to release a new version of Beaker with this fix. We just finalizing a new release process.

closed 2024-01-02 by StykMartin

jonmasters, 2024-01-02:
Thanks! It took me a few hours to find it as I’m not a beaker developer so
I had to learn as I went. But the good news is hey I learned a lot about
hacking on the code. Hope you’re all doing great!

Computer Architect