beaker-project/beaker #208: [BUG] downloadable PDF documentation is missing

pbunyan, 2024-01-30:
**Describe the bug**  
Downloadable PDF documentation is missing

https://beaker-project.org/docs/user-guide.pdf
404 Not Found
Not Found
The requested URL /docs/user-guide.pdf was not found on this server.

https://beaker-project.org/docs/admin-guide.pdf
404 Not Found
Not Found
The requested URL /docs/admin-guide.pdf was not found on this server.

**To Reproduce**  
Steps to reproduce the behavior:
1. Go to '...'
https://beaker-project.org/docs/user-guide.pdf
https://beaker-project.org/docs/admin-guide.pdf

**Actual behavior**  
The PDFs are missing

**Expected behavior**  
Access to the Beaker documentation in PDF format

**Screenshots**  
If applicable, add screenshots to help explain your problem.

**Additional context**  
We can reach the documentation here
https://beaker-project.org/docs/user-guide/
https://beaker-project.org/docs/admin-guide/

But scrolling/clicking thru the html pages is not user friendly.
Please bring back the PDF format :)

StykMartin, 2024-01-30:
I will try to rebuild it :)

StykMartin, 2024-01-30:
@pbunyan, since we don't have this for Release 29 at this moment, you can use 
https://beaker-project.org/docs-release-28/user-guide.pdf
https://beaker-project.org/docs-release-28/admin-guide.pdf

It is quite similar, except for the alembic, of course.

pbunyan, 2024-01-30:
Its great to know the Beaker God is still out there :)
The PDF documentation has always been a great resource for the Beaker users.
Thank you for keeping it up to date and available.

Thank you!  :)

StykMartin, 2024-02-11:
And it is done.
[image]

https://beaker-project.org/docs/user-guide.pdf
https://beaker-project.org/docs/admin-guide.pdf

closed 2024-02-11 by StykMartin

pbunyan, 2024-02-12:
Excellent!
Thank you Martin!