beaker-project/beaker #211: [BUG]SyntaxWarning: invalid escape sequence on python3.12

HouMinXi, 2024-02-05:
**Describe the bug**  
when use bkr submit a job, an error occurred as below:
Submitted: ['J:8890767']
/usr/lib/python3.12/site-packages/bkr/client/__init__.py:65: SyntaxWarning: invalid escape sequence '\w'
  matched = re.match('^(\w+)\s+(\S+.*)$', line)
/usr/lib/python3.12/site-packages/bkr/client/commands/cmd_workflow_xslt.py:38: SyntaxWarning: invalid escape sequence '\['
  self.re_tag = re.compile('(.*)(\[@(.*)=(.*)\])')
**Version-Release number**  

**To Reproduce**  
Steps to reproduce the behavior:
1. job xml as below
<job retention_tag="active+1" product="cpe:/o:redhat:enterprise_linux" user="">
    <whiteboard><![CDATA[rhel9.4 ctc2 mlx5_core ovs_bond_function dpdk RHEL-9.4.0-20240205.8 rtk openvswitch3.2-3.2.0-40.el9fdp]]></whiteboard>
    <notify>
    </notify>
    <recipeSet priority="Normal">
        <recipe kernel_options="" kernel_options_post="pci=realloc intel_iommu=on" whiteboard="rhel9.4 ctc2 mlx5_core ovs_bond_function dpdk RHEL-9.4.0-20240205.8 rtk openvswitch3.2-3.2.0-40.el9fdp" ks_meta="harness='restraint-rhts beakerlib' redhat_ca_cert method=nfs no_networks redhat_ca_cert disabled_root_access">
            <autopick random="false" />
            <distroRequires>
                <and>
                    <distro_method op="=" value="nfs" />
                    <distro_name op="=" value="RHEL-9.4.0-20240205.8" />
                    <distro_variant op="=" value="BaseOS" />
                    <distro_arch op="=" value="x86_64" />
                </and>
                <or>
                </or>
                <not>
                    <or>
                    </or>
                </not>
            </distroRequires>
            <hostRequires>
                <and>
                    <hostname op="like" value="dell-per740-15.rhts.eng.pek2.redhat.com" />
                </and>
                <system_type op="=" value="Prototype" />
            </hostRequires>
            <repos>
            </repos>
            <partitions>
            </partitions>
            <ks_appends>
                <ks_append><![CDATA[
%post --log=/root/my-ks-post.log

downhostname=download.devel.redhat.com

#generate buildroot repo
rtype=rel-eng
rtype=nightly
verx=$(rpm -E %rhel)
case $verx in
8|9)
	buildrootUrl=http://$downhostname/rhel-$verx/$rtype/BUILDROOT-$verx/latest-BUILDROOT-$verx-RHEL-$verx/compose/Buildroot/$(arch)/os/
	cat <<-EOF >/etc/yum.repos.d/beaker-buildroot.repo
	[beaker-buildroot]
	name=beaker-buildroot
	baseurl=$buildrootUrl
	enabled=1
	gpgcheck=0
	skip_if_unavailable=1
	EOF
	;;
esac

#update restraint plugins
_rpath=share/restraint/plugins/task_run.d
_path=qa/rhts/lookaside/bkr-client-improved/$_rpath
[... 158 lines cut ...]
        </recipe>
    </recipeSet>
</job>

2. bkr job-submit test.xml
/usr/lib/python3.12/site-packages/bkr/client/__init__.py:64: SyntaxWarning: invalid escape sequence '\w'
  matched = re.match('^(\w+)\s+(\S+.*)$', line)
/usr/lib/python3.12/site-packages/bkr/client/commands/cmd_workflow_xslt.py:38: SyntaxWarning: invalid escape sequence '\['
  self.re_tag = re.compile('(.*)(\[@(.*)=(.*)\])')

**Actual behavior**  
/usr/lib/python3.12/site-packages/bkr/client/__init__.py:64: SyntaxWarning: invalid escape sequence '\w'
  matched = re.match('^(\w+)\s+(\S+.*)$', line)
/usr/lib/python3.12/site-packages/bkr/client/commands/cmd_workflow_xslt.py:38: SyntaxWarning: invalid escape sequence '\['
  self.re_tag = re.compile('(.*)(\[@(.*)=(.*)\])')

**Expected behavior**  
no error message printed. 

**Additional context**  
$ rpm -qa | grep beaker*
beakerlib-1.29.3-3.fc39.noarch
python3-beaker-1.12.1-3.fc39.noarch
beaker-common-29.0-1.fc39.noarch
beaker-client-29.0-1.fc39.noarch

mh21, 2024-02-05:
Very ugly CLI invocation to find more candidates:

```
git grep '\<re\..*(["'"'].*\\\\" | grep '\.py:' | sed 's/^\([^ ]*\):\(.*\)/- [`\1`](..\/tree\/python-3\/\1): `\2`/g'
```

Results:
- [`Client/src/bkr/client/__init__.py`](../tree/python-3/Client/src/bkr/client/__init__.py): `                matched = re.match('^(\w+)\s+(\S+.*)$', line)`
- [`Client/src/bkr/client/commands/cmd_workflow_xslt.py`](../tree/python-3/Client/src/bkr/client/commands/cmd_workflow_xslt.py): `        self.re_tag = re.compile('(.*)(\[@(.*)=(.*)\])')`
- [`Client/src/bkr/client/wizard.py`](../tree/python-3/Client/src/bkr/client/wizard.py): `RegExpBug        = re.compile("^\d+$")`
- [`Client/src/bkr/client/wizard.py`](../tree/python-3/Client/src/bkr/client/wizard.py): `RegExpBugLong    = re.compile("^bz\d+$")`
- [`Client/src/bkr/client/wizard.py`](../tree/python-3/Client/src/bkr/client/wizard.py): `RegExpCVE        = re.compile("^\d{4}-\d{4,}$")`
- [`Client/src/bkr/client/wizard.py`](../tree/python-3/Client/src/bkr/client/wizard.py): `RegExpCVELong    = re.compile("^CVE-\d{4}-\d{4}$")`
- [`Client/src/bkr/client/wizard.py`](../tree/python-3/Client/src/bkr/client/wizard.py): `RegExpScript     = re.compile("\.(sh|py|pl)$")`
- [`Client/src/bkr/client/wizard.py`](../tree/python-3/Client/src/bkr/client/wizard.py): `RegExpMetadata   = re.compile("(\$\(METADATA\):\s+Makefile.*)$", re.S)`
- [`Client/src/bkr/client/wizard.py`](../tree/python-3/Client/src/bkr/client/wizard.py): `RegExpTest       = re.compile("TEST=(\S+)", re.S)`
- [`Client/src/bkr/client/wizard.py`](../tree/python-3/Client/src/bkr/client/wizard.py): `RegExpVersion    = re.compile("TESTVERSION=([\d.]+)", re.S)`
- [`Client/src/bkr/client/wizard.py`](../tree/python-3/Client/src/bkr/client/wizard.py): `GuessAuthorDomain = re.sub("^.*\.([^.]+\.[^.]+)$", "\\1", os.uname()[1])`
- [`Client/src/bkr/client/wizard.py`](../tree/python-3/Client/src/bkr/client/wizard.py): `    text = re.compile("\s+").sub(" ", text)`
- [`Client/src/bkr/client/wizard.py`](../tree/python-3/Client/src/bkr/client/wizard.py): `    return re.compile("\n" + " " * count).sub("\n", text)`
- [`Client/src/bkr/client/wizard.py`](../tree/python-3/Client/src/bkr/client/wizard.py): `    return re.compile("\n").sub("\n" + " " * count, text)`
- [`Client/src/bkr/client/wizard.py`](../tree/python-3/Client/src/bkr/client/wizard.py): `    result = re.compile("\s+$", re.M).sub("", result)`
- [`Client/src/bkr/client/wizard.py`](../tree/python-3/Client/src/bkr/client/wizard.py): `            return re.sub("\n\s+$", "", content.firstChild.nodeValue)`
- [`Client/src/bkr/client/wizard.py`](../tree/python-3/Client/src/bkr/client/wizard.py): `                m = re.search("%s=(\S+)" % var, self.text)`
- [`Client/src/bkr/client/wizard.py`](../tree/python-3/Client/src/bkr/client/wizard.py): `                if m: self.text = re.sub("\$\(%s\)" % var, m.group(1), self.text)`
- [`Client/src/bkr/client/wizard.py`](../tree/python-3/Client/src/bkr/client/wizard.py): `            m = re.search("echo\s+[\"'](\w+):\s*(.*)[\"']", line)`
- [`Client/src/bkr/client/wizard.py`](../tree/python-3/Client/src/bkr/client/wizard.py): `            value = re.sub("\\\\\"", "\"", m.group(2))`
- [`Client/src/bkr/client/wizard.py`](../tree/python-3/Client/src/bkr/client/wizard.py): `        m = re.search("(.*)\s+<(.*)>", options.opt.owner)`
- [`Client/src/bkr/client/wizard.py`](../tree/python-3/Client/src/bkr/client/wizard.py): `        self.text = re.sub("\n\n\trhts-lint",`
- [`Client/src/bkr/client/wizard.py`](../tree/python-3/Client/src/bkr/client/wizard.py): `        self.data = re.sub("^\s*", "", self.data)`
- [`Client/src/bkr/client/wizard.py`](../tree/python-3/Client/src/bkr/client/wizard.py): `        self.data = re.sub("\s*$", "", self.data)`
- [`Client/src/bkr/client/wizard.py`](../tree/python-3/Client/src/bkr/client/wizard.py): `        self.data = re.sub("\s+", " ", self.data)`
- [`Client/src/bkr/client/wizard.py`](../tree/python-3/Client/src/bkr/client/wizard.py): `            m = re.search("^\+\S+(.*)", answer)`
- [`Client/src/bkr/client/wizard.py`](../tree/python-3/Client/src/bkr/client/wizard.py): `        text = re.sub("[^\w\s]", "", text)`
- [`Client/src/bkr/client/wizard.py`](../tree/python-3/Client/src/bkr/client/wizard.py): `        m = re.match("^(\d{1,2})[mhd]$", self.data)`
- [`Client/src/bkr/client/wizard.py`](../tree/python-3/Client/src/bkr/client/wizard.py): `        return re.match("^\d+\.\d+$", self.data)`
- [`Client/src/bkr/client/wizard.py`](../tree/python-3/Client/src/bkr/client/wizard.py): `            return re.sub("CVE-\d{4}-\d{4}\s*", "", removeEmbargo(self.bug.summary))`
- [`Client/src/bkr/client/wizard.py`](../tree/python-3/Client/src/bkr/client/wizard.py): `        skeleton = re.sub("\n\s+$", "\n", skeleton)`
- [`IntegrationTests/src/bkr/inttest/client/test_update_inventory.py`](../tree/python-3/IntegrationTests/src/bkr/inttest/client/test_update_inventory.py): `        m = re.search('J:(\d+)', out)`
- [`IntegrationTests/src/bkr/inttest/client/test_watchdogs_show.py`](../tree/python-3/IntegrationTests/src/bkr/inttest/client/test_watchdogs_show.py): `        self.assertTrue(re.match('%s: \d\d\\n' % t1.id, out))`
- [`IntegrationTests/src/bkr/inttest/client/test_workflow_simple.py`](../tree/python-3/IntegrationTests/src/bkr/inttest/client/test_workflow_simple.py): `        m = re.search('J:(\d+)', out)`
[... cut ...]
- [`Server/bkr/server/search_utility.py`](../tree/python-3/Server/bkr/server/search_utility.py): `        wildcard_y = re.sub('\*','%',y)`
- [`Server/bkr/server/search_utility.py`](../tree/python-3/Server/bkr/server/search_utility.py): `        wildcard_y = re.sub('\*','%',y)`
- [`Server/bkr/server/testinfo.py`](../tree/python-3/Server/bkr/server/testinfo.py): `        m = re.match('^(\d+)(.*)$', value)`
- [`Server/bkr/server/testinfo.py`](../tree/python-3/Server/bkr/server/testinfo.py): `            m = re.match('([^:]*)\((.*)\):(.*)', line)`

StykMartin, 2024-02-05:
@mh21 If you file a PR I will backport it 👀

closed 2024-02-09 by StykMartin