beaker-project/beaker #255: [RFE]

spinegar1, 2025-04-22:
**Is your feature request related to a problem? Please describe.**
Creating a new Feature to encompass the work talked about in t[his previously approved feature request](https://github.com/beaker-project/beaker/issues/179).

I would like to modernize the Beaker code by removing the TurboGears/CherryPy dependencies and replacing them with the popular Flask/Gunicorn standard so more people can contribute to and understand the code. This will also be an opportunity to ensure all dependent packages work with Python3.10+ to enable security support through 2026 and RHEL9 compatibility. 

**Describe the solution you'd like**
Remove TurboGears references from the Server & Update Associated Integration Tests;
* TurboGears Functionality to be replaced by Flask/SQLAlchemy
** Database -> SQLAlchemy built-in
** Config -> Flask Config Env Vars
** Widgets -> Flask WTF Forms
** Flashes -> Flask Templating
* Update Build (setup.py & .spec files)
** Include updated Flask/SQLAlchemy packages
** Remove TurboGears and dependencies

Remove all CherryPy endpoints
Use Standard RestFUL API Schema
CherryPy XMLRPC Endpoint Replacement
CherryPy Endpoint Functionality to be replaced by Flask to reduce number of web frameworks in Beaker; This work to be done CherryPy Class/File by Class/File (LabController, Jobs, Distro, etc)
Request -> Flask Request
Response -> Flask Response
Expose Endpoint to URL -> Flask URL attached to endpoint
Errors -> Flask Error Handling
Serve Files -> Flask File Serving
Grouping API Endpoints  -> Flask Blueprints
Leave CherryPy to serve Flask app in meantime to reduce surface area of change.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

p3ck, 2025-04-23:
> --- **Other Considerations** ---
> 
> While Flask does support XML templates, they are not as common as HTML templates; Upgrading the Beaker project to use HTML Templates that are formatted via Jinja2 instead of using XML Templates formatted via Kid would be another step towards using tools that are well known, documented, and popular in the OpenSource community. This piece of work can be done at a future date because Flask can format XML files so it wont be a blocker.
> 

Which XML templates are you talking about? 

Beaker jobs are described in XML but we could move to yaml, but there are a lot of jobs/templates outside of beaker that users have crafted.  I think we would need to support XML as well.

The server web templates are in turbogears .kid which could be migrated to jinja2.

p3ck, 2025-04-23:
I must not have had enough coffee..  I see now you are talking about the kid templates.  I don't see any issue in moving to jinja2 here.  but the job xml would need to be backwards supported I believe.

spinegar1, 2025-05-29:
Sounds good. Flask is able to support XML but doesn't seem to support the KID templating engine so the XML templates would need to be updated to use Jinja2 - hopefully this isn't to big of a lift. 

Handling the dynamic widgets and flash's that are supported by TurboGears currently will be a bit harder; the lowest lift for this will probably be replacing the Turbogears widgets and flash components with custom  Flask widgets/flashes with a library like [Flask WTF](https://flask-wtf.readthedocs.io/en/1.2.x/). 

The templates will need to be reworked to dynamically change the widget/flash  used at the template level via arguments passed in via the view instead of the view handling it by dynamically generating and injecting the flash/widget into the template (how TurboGears currently handles these). 

This widget/flash dynamic injection happens fairly frequently in the current TurboGears code and adjusting it to work for Flask will require a fairly significant lift. 

TurboGears2 seems to support widget/flash dynamic injection at the view level and thats what made me think it might be less of a lift, but @StykMartin brought up that using TurboGears2 would require us to rewrite the templates anyways, could you help me understand why using TurboGears2 would require rewriting the templates?